WMT-GmbH/pn532 #1: Documentation examples are using x.ms() methods

plaes, 2021-11-26:
When attempting to use the examples, code like this:
`pn532.process(&Request::INLIST_ONE_ISO_A_TARGET, 7, 1000.ms());`

Causes errors:
`can't call method `ms` on ambiguous numeric type `{integer}``

Should these changed to use `std::time::Duration::from_millis()` instead?

dimpolo, 2021-11-26:
Good point.

Most HAL's I've worked with define an extension trait on `u32` in their prelude (e.g. [stm32f4xx-hal](https://docs.rs/stm32f4xx-hal/0.10.1/stm32f4xx_hal/time/trait.U32Ext.html))

The concrete type of the timeout value will depend on the implementation of [CountDown](https://docs.rs/embedded-hal/0.2.6/embedded_hal/timer/trait.CountDown.html).

`core::time::Duration` would be one possible type but often (especially with `no-std`) other types are used.

I would like to keep the `.ms()` call since it doesn't mention a concrete type, but maybe we could provide our own extension trait such as:

```rust
use core::time::Duration;

pub trait U64Ext {
    fn ms(self) -> Duration;
}

impl U64Ext for u64 {
    fn ms(self) -> Duration {
        Duration::from_millis(self)
    }
}
```

Would that work for you?

plaes, 2021-11-26:
Yeah, I was trying to use it without HAL (or with basic linux-embedded-hal) and couldn't figure out where to import these definitions (see issue #2 ).

commit e7a029c4 (2021-11-26): fixes #1
closed 2021-11-26 by dimpolo