WMT-GmbH/pn532 #4: InterfaceError if wait_ready sent immediately after send or receive_ack

anowell, 2022-11-12:
I've been tinkering with the PN532 in [this Grove NFC module](https://wiki.seeedstudio.com/Grove_NFC/) in I2C configuration attached to a Raspberry Pi Pico. For me, calling `process` (e.g. `pn532.process(&Request::GET_FIRMWARE_VERSION, 4, 100.millis())`) consistently results in `InterfaceError(Abort(1))` (which I believe is `ABRT_7B_ADDR_NOACK` from the RP2040 datasheet - no ack from address). After a bit of debugging, I've narrowed it down to the `wait_ready` calls happening too soon after `send` or `receive_ack`. If I delay the `wait_ready` by 1ms or simply retry `wait_ready` after the `InterfaceError(Abort(1))`, things seem to work as expected.

I'm pretty new to working with I2C, so it's not clear to me if that's something this library should handle/tolerate. I vaguely recall [this PN532 C lib](https://github.com/Seeed-Studio/PN532/) not giving me this trouble (though it was a while back with a different board). At quick glance, they seem to [retry the wait for ack until timeout without any other error handling or early return conditions](https://github.com/Seeed-Studio/PN532/blob/40fa7418636da5e03b0126e38c221b143b24b5a4/PN532_I2C/PN532_I2C.cpp#L219-L236). If it makes sense to you to continue retrying on `InterfaceError(_)`, I'd be willing to put together a PR.

I'm gonna switch to using the IRQ interface, as I suspect it will make this a non-issue, but I figured I'd at least point this out. I also have a different PN532 device I'll try soon.

anowell, 2022-11-12:
As suspected, this is not an issue when using `I2CInterfaceWithIrq`.

dimpolo, 2022-11-12:
Thanks for investigating this!

> If it makes sense to you to continue retrying on InterfaceError(_), I'd be willing to put together a PR.

This might be a good way forward but before I'd like to better understand why the error occurs.

It would be interesting to know if this has something to do with the Grove NFC module, the RP2040 Hardware or the RP2040 software.
What baud rate is your I2C bus running at?
Do you have access to an oscilloscope or logic analyzer and would you be able to post the waveforms here?

anowell, 2022-11-12:
I2C was at 400 KHz. I did also try 100KHz with the same result.

I don't have a scope, but I did just have a couple cheap logic analyzers arrive yesterday - I just need a bit of time to teach myself how to use them, but I'll post the waveforms once I figure it out.

I also have an ITEAD PN532 NFC module that I'll test on; I should be able to quickly swap that out tomorrow just to compare different PN532 modules on the same RP2040.

anowell, 2022-11-12:
Fwiw, I changed some of the wiring connecting the PN532 module (previous wiring was a bit suspect), and now it only repros in `--release` mode, but I get the same behavior with both the the Grove - NFC and ITEAD PN532 modules. 

<details>
  <summary>Code snippet</summary>

Sends GET_FIRMWARE_VERSION twice. First with logic to retry on `wait_ready` errors, and then via `process`.

  ```rust
    info!("Sending GET_FIRMWARE_VERSION");
    if let Err(err) = pn532.send(&Request::GET_FIRMWARE_VERSION) {
        error!("Error sending GET_FIRMWARE_VERSION: {}", defmt::Debug2Format(&err));
    }

    info!("Wait Ready");
    for _ in 0..10 {
        match pn532.interface.wait_ready() {
            Poll::Ready(Ok(_)) => {
                info!("Receive ACK");
                let ack_res = pn532.receive_ack();
                info!("ACK Result: {}", defmt::Debug2Format(&ack_res));
                break;
            }
            Poll::Pending => {
                info!("Not ready to receive ack");
            }
            Poll::Ready(Err(err)) => {
                error!("Error waiting for ready: {}", defmt::Debug2Format(&err));
            }
        };
    }

    info!("Wait Ready");
    for _ in 0..10 {
        match pn532.interface.wait_ready() {
            Poll::Ready(Ok(_)) => {
                let result = pn532.receive_response(Request::GET_FIRMWARE_VERSION.command, 4);
                info!("GET_FIRMWARE_VERSION response: {:?}", defmt::Debug2Format(&result));
                break;
            }
            Poll::Pending => {
                info!("Not ready to receive ack");
                delay.delay_ms(1);
            }
            Poll::Ready(Err(err)) => {
                error!("Error waiting for ready: {}", defmt::Debug2Format(&err))
            }
        };
    }

    let result = pn532.process(&Request::GET_FIRMWARE_VERSION, 4, 200.millis());
    info!("GET_FIRMWARE_VERSION process: {:?}", defmt::Debug2Format(&result));
  ```

</details>

<details>
  <summary>Output</summary>

#### Debug output

  ```
INFO  Sending GET_FIRMWARE_VERSION
INFO  Wait Ready
INFO  Receive ACK
INFO  ACK Result: Ok(())
INFO  Wait Ready
INFO  GET_FIRMWARE_VERSION response: "Ok([50, 1, 6, 7])"
INFO  GET_FIRMWARE_VERSION process: "Ok([50, 1, 6, 7])"
  ```

#### Release output

  ```
INFO  Sending GET_FIRMWARE_VERSION
INFO  Wait Ready
ERROR Error waiting for ready: Abort(1)
INFO  Receive ACK
INFO  ACK Result: Ok(())
INFO  Wait Ready
INFO  Not ready to receive ack
INFO  GET_FIRMWARE_VERSION response: "Ok([50, 1, 6, 7])"
INFO  GET_FIRMWARE_VERSION process: "Err(InterfaceError(Abort(1)))"
  ```

</details>

Next up is to figure out this logic analyzer. :smile:

dimpolo, 2023-03-06:
Did you happen to find out more?

anowell, 2023-03-06:
I have not. It's still on my to do list, but not very urgent, and I've yet to break in my logic analyzers.

sebarnd, 2023-07-09:
I think I've figured it out: while waiting to be ready the PN532 doesn't at all answer to requests for its address. `i2c.read()` in `wait_ready` consequently returns an error and the whole thing stops. Adding `let _ = ` solves the issue for me. I'm quite new to rust so i don't really feel comfortable with submitting a patch but I'm happy to test....

dimpolo, 2023-07-13:
I had a look at this with a logic analyzer.
The protocol looks like this:
* send request
* wait 1
* read ack
* wait 2
* read response

If one sends a I2C message during "wait 1" and the PN532 is not ready it will not ACK the read request in `Interface::wait_ready`
[image]

Interestingly this is not the case for "wait 2", where things work as expected.

The way I see it this leaves us two options:
* add a delay after "send request"
* ignore I2C errors in `Interface::wait_ready`

I went with the second option in #14 but I'd be interested in what you think @anowell @sebarnd

sebarnd, 2023-07-13:
As the device has already answered before I think ignoring protocol errors here shouldn't be too harmful. But taking also in account that the bus needs to be constantly polled by this approach it seems more elegant to have some delay there.

dimpolo, 2023-07-13:
Those are some good points.
Running I2C at 400kHz there's up to 10 reads that fail before succeeding. This might be different for other commands than GET_FIRMWARE_VERSION.

I found this in the PN532 User Manual `6.2.4`:

> it may happen that the PN532 does not acknowledge its own address immediately after having finished a previous exchange.

Another thing to take into account is that a delay doesn't make much sense for the other interfaces or even for some of the other provided public API like `Pn532::process_async`

sebarnd, 2023-07-14:
Maybe there could be a delay after a AdrNACK'ed communication? So that there would be somehow a rate limiting for the polls?

dimpolo, 2023-07-17:
After thinking about this a bit I've decided with the following:
- in the short term go with the quick fix in #14
- in the long term, redesign the API in a way where where we can utilize the provided timer to handle this case in a better way

Thanks for your help

closed 2023-07-17 by dimpolo

liebman, 2024-05-05:
Just adding this as it seems relevant: PN532 datasheet section "12.26 Timing for the I2C interface" lists a minimum time between a STOP condition and a START condition of 1.3ms...  It also states that it may need to stretch the clock when woken by up to 1ms.

PR #14 (2023-07-17): fix `I2CInterface` (without Irq pin) `InterfaceError`
fixes #4