WMT-GmbH/pn532 #5: Fails to build when used in a esp32 project with feature "std"

JohnTheCoolingFan, 2022-12-25:
Cargo project generated using template from https://github.com/esp-rs/esp-idf-template
Added `pn532 = { version = "0.2", features = ["std"] }` to dependencies.
Full error log: https://gist.github.com/JohnTheCoolingFan/ca330da3b7e187e28124d98f0573abba

dimpolo, 2022-12-25:
Thanks for the report!

The `std` feature pulls in the `serialport` crate, which only compiles for some [targets](https://github.com/serialport/serialport-rs#platform-support).

Currently that is all that the `std` feature does, so there is no reason to enable it for the esp32. 
I'll try and document this better 😃

JohnTheCoolingFan, 2022-12-25:
Alright, thanks for clarifying

dimpolo, 2023-01-29:
should be documented now

closed 2023-01-29 by dimpolo