WMT-GmbH/pn532 #6: How to use with esp-idf-hal?

JohnTheCoolingFan, 2022-12-25:
esp-idf-hal is implementation of embedded-hal for esp32. I cannot find any type that implements i2c::Transactional from embedded_hal. Am I missing something?

dimpolo, 2022-12-25:
Thanks for the report! 

I looked around the `esp-idf-hal` a bit, you're not missing anything 😃.

Basically we have three options here:

### The one where I do the work:
[I2CDriver](https://esp-rs.github.io/esp-idf-hal/esp_idf_hal/i2c/struct.I2cDriver.html) does have an impl of the `embedded-hal v1.0.0` version of the [I2c](https://docs.rs/embedded-hal/1.0.0-alpha.9/embedded_hal/i2c/trait.I2c.html) trait and that trait includes a `transaction` method. 
We could add a `eh1` feature flag `I2CInterface` in terms of the new trait.

I definitely want to make this happen but I won't get to this until in a couple of weeks. (I would of course welcome a PR 😉)

### The one where the esp crew does the work:
You could open an issue over there asking for an implementation of `Transactional` (or even write a PR yourself). They have been very helpful and open to suggestions in my experience.

### The one where you do the work:
You can implement `Interface` yourself on a wrapper type of [I2CDriver](https://esp-rs.github.io/esp-idf-hal/esp_idf_hal/i2c/struct.I2cDriver.html). 
``` rust
struct I2CWrapper<'a>(I2CDriver<'a>);

impl<'a> pn532::Interface for I2CWrapper<'a> {
    ...
}
```
You can have a look at https://github.com/WMT-GmbH/pn532/blob/master/src/i2c.rs#L30-L60. I'll be glad to help you out if questions come up.

JohnTheCoolingFan, 2022-12-25:
Thank you for such a great reply! I'll report it to esp crew and maybe do a PR, as I think that's an error from their side.

liebman, 2024-04-28:
Will be fixed "soon-ish": https://github.com/esp-rs/esp-hal/pull/1505

dimpolo, 2024-04-28:
Thanks for your work!

LimesKey, 2024-05-04:
I'm having some issues doing this fix,
```rust
    let peripherals = Peripherals::take().unwrap();
    let sda = peripherals.pins.gpio14;
    let scl = peripherals.pins.gpio13;
    let config = I2cConfig::new().baudrate(esp_idf_svc::hal::prelude::Hertz(50000));
    let i2cdriver = I2cDriver::new(peripherals.i2c0, sda, scl, &config).unwrap();

    struct I2CWrapper<'a>(I2cDriver<'a>);

    impl<'a> pn532::Interface for I2CWrapper<'a> {
        type Error = I2CWrapper::Error;

        fn write(&mut self, frame: &[u8]) -> Result<(), Self::Error> {
            I2CWrapper::write(self, frame)
        }
    
        fn wait_ready(&mut self) -> Poll<Result<(), Self::Error>> {
            I2CWrapper::wait_ready(self)
        }
    
        fn read(&mut self, buf: &mut [u8]) -> Result<(), Self::Error> {
            I2CWrapper::read(self, buf)
        }
    }

    let mut pn532 = Pn532::new(i2cdriver, config);
    ```
    `missing lifetime specifier` and `ambiguous associated` for the `type Error` line....

liebman, 2024-05-05:
Pn532 needs to implement for `embedded-hal` 1.0 traits.

dimpolo, 2024-12-15:
Should be fixed by #22

closed 2024-12-15 by dimpolo