WMT-GmbH/pn532 #13: TimeoutAck error when using library with rppal

kelada, 2023-02-26:
Hi there. I'm interested in using this library to replace an existing application I've written using https://github.com/adafruit/Adafruit-PN532 python library. I am new to embedded rust so it's very likely I'm doing something wrong.

Running the example below I always get `TimeoutAck` errors, but I'm not sure why. Looking at the Adafruit library I've understood that I need to enable the `msb-spi` flag in addition to matching the clock frequency ([source](https://github.com/adafruit/Adafruit-PN532/blob/22e21a303fb9b36ec0c52c3d8f449dc0b4477576/Adafruit_PN532.cpp#L92-L97))

However, I don't seem to have had much luck with this. Any guidance would be greatly appreciated.

#### Hardware:
- Raspberry Pi 3 B
- NFC Module v3 (can be found on amazon if you search pn532, red rectangular PCB)

#### Software
`Cargo.toml`
```
...
[dependencies]
pn532 = { version = "0.2.2", features = ["msb-spi"] }
rppal = { version = "0.14.1", features = ["hal"] }
```

`main.rs`
```
use pn532::{spi::SPIInterface, IntoDuration, Pn532, Request};
use rppal::gpio::Gpio;
use rppal::hal::Timer;
use rppal::spi::{Bus, Mode, SlaveSelect, Spi};
use std::error::Error;

/// RPi 3B
// SPI: SPI0 (GPIO 10, 9, 11)
// CS:  GPIO 8
fn main() -> Result<(), Box<dyn Error>> {
    let timer = Timer::new();

    let spi = Spi::new(Bus::Spi0, SlaveSelect::Ss0, 1_000_000, Mode::Mode0)?;
    let cs = Gpio::new()?.get(8)?.into_output();
    let interface = SPIInterface { spi, cs };

    let mut pn532: Pn532<_, _, 32> = Pn532::new(interface, timer);
    if let Ok(uid) = pn532.process(&Request::INLIST_ONE_ISO_A_TARGET, 7, 1000.ms()) {
        let result = pn532.process(&Request::ntag_read(10), 17, 50.ms()).unwrap();
        if result[0] == 0x00 {
            println!("page 10: {:?}", &result[1..5]);
        }
    } else {
        println!("Found: nothing");
    }

    Ok(())
}

```

dimpolo, 2023-02-26:
Hi. Thanks for taking the time to submit this issue. There is currently a bug SPI implementation in 0.2.2 which is fixed by #7. 

I haven't gotten around publishing a new version, sorry about that.

Could you check if using this fixes your problem?
```toml
pn532 = { git = "https://github.com/WMT-GmbH/pn532", features = ["msb-spi"] }
```

dimpolo, 2023-02-26:
You'll probably also want to include this line before inlisting the target (I should update the readme)

```rust
pn532.process(
    &Request::sam_configuration(SAMMode::Normal, false),
    0,
    50.ms(),
)?;
```

kelada, 2023-02-26:
Thanks for the quick response.

Unfortunately it throws an `TimeoutAck` error at this new line.

dimpolo, 2023-02-26:
> Unfortunately it throws an `TimeoutAck` error at this new line.

That is with the updated git dependency?

kelada, 2023-02-26:
> > Unfortunately it throws an `TimeoutAck` error at this new line.
> 
> That is with the updated git dependency?

Yeah, I still get the same error.

```
[dependencies]
pn532 = { git = "https://github.com/WMT-GmbH/pn532", branch = "master", features = ["msb-spi"] }
rppal = { version = "0.14.1", features = ["hal"] }
```

```
thread 'main' panicked at 'called `Result::unwrap()` on an `Err` value: TimeoutAck', src/main.rs:25:10
```

Note: I'm using `unwrap()` rather than `?` to get it to panic if there is an error.

dimpolo, 2023-02-26:
Hm strange. At this point I can't see why the code woudn't work. 

Could you compare the signals with the adafruit implementation using a scope or logic analyzer?

I'll try and reproduce the issue on my end but I won't have access to the hardware until in two weeks.

kelada, 2023-02-26:
>Could you compare the signals with the adafruit implementation using a scope or logic analyzer?

I don't have either on hand today, i'll see if I can get ahold of one this week.

In the meantime, do you think it's worth me testing I2C to see that that works?

dimpolo, 2023-02-26:
Sure thing! Be sure to cross check with the adafruit library to rule out hardware issues :)

kelada, 2023-03-02:
In regard to SPI, changing the CS pin from 8 to 5 on the RPi seems to have resolved the consistant `TimeoutAck` error when calling `sam_configuration(...)`. I'm satisfied that this is not an issue with the library. Thanks for you help.

closed 2023-03-02 by kelada

dimpolo, 2023-03-02:
Thanks for checking back in. Glad you found a solution :)

Clemsim, 2025-06-05:
Hi, i same exact problem with a Pi Zero and a PN532 that i bought on amazon.
But when i run my code, i get TimeOutAck errors, with and without the msb-spi feature activated.
Do you have a real solution, apart from changing the cs pin?
`use std::convert::Infallible;
use std::time::Duration;
use std::time::Instant;

use pn532::nb;
use pn532::CountDown;
use rppal::spi::{Spi, Bus, SlaveSelect, Mode};
use rppal::spi::SimpleHalSpiDevice;
use pn532::Pn532;
use pn532::Request;
use pn532::requests::SAMMode;
use pn532::IntoDuration;

// Stolen from rppal's implementation
#[derive(Debug, Copy, Clone)]
struct Timer {
    start: Instant,
    duration: Duration,
}

impl Timer {
    pub fn new() -> Self {
        Self {
            start: Instant::now(),
            duration: Duration::from_micros(0),
        }
    }
}

impl Default for Timer {
    fn default() -> Self {
        Timer::new()
    }
}

impl CountDown for Timer {
    type Time = Duration;

    fn start<T>(&mut self, timeout: T)
    where
        T: Into<Self::Time>,
    {
        self.start = Instant::now();
        self.duration = timeout.into();
    }

    fn wait(&mut self) -> nb::Result<(), Infallible> {
        if self.start.elapsed() >= self.duration {
            Ok(())
        } else {
            Err(nb::Error::WouldBlock)
        }
    }
}
fn main() {
    let spi_bus = Spi::new(Bus::Spi0, SlaveSelect::Ss0, 1_000_000, Mode::Mode0)
        .expect("Couldn't create SPI bus");
    let spi_device = SimpleHalSpiDevice::new(spi_bus);
    let timer = Timer::new();
    let spi_interface = pn532::spi::SPIInterface{ spi: spi_device };
    let mut pn532: Pn532<_, _, > = Pn532::new(spi_interface, timer);

    println!("Attempting to get firmware version...");
    match pn532.process(&Request::GET_FIRMWARE_VERSION, 4, 500.ms()) {
        Ok(firmware) => {
            println!("PN532 Firmware version: {:?}", firmware);
        }
        Err(e) => {
            println!("Error getting firmware version: {:?}", e);
            return;
        }
    }

    println!("Attempting SAM configuration...");
    // SAM configuration expects 0 data bytes in the response payload
    match pn532.process(&Request::sam_configuration(SAMMode::Normal, false), 0, 500.ms()) {
        Ok(_) => { // The actual response data is empty for SAM_configuration success
            println!("PN532 SAM_configuration successful!");
        }
        Err(e) => {
            println!("Error during SAM_configuration: {:?}", e);
            return;
        }
    }
    println!("PN532 initialized. (End of simplified test)");
}`

dimpolo, 2025-06-11:
Do you have a way to test, if you're facing a hardware issue?
Use a logic analyzer or try a different library and see if anything changes?

Clemsim, 2025-06-11:
I dont know of any other libraries to do pn532 connection, but i have a small script python that works.
However, this python script works with /dev/spidev not showing, so i think it works with software spi.