WMT-GmbH/pn532 #15: `requests::Command` needs better documentation for it's variants

JohnTheCoolingFan, 2024-03-03:
Currently the doc comments refer to some document I assume, which is not a very good practive. I think it should contain an explanation, a description of the command variant.

JohnTheCoolingFan, 2024-03-03:
I might do a PR on this because the hardest part (for me) would probably be finding the document that is being referred to here. No promises.

dimpolo, 2024-03-04:
I get your point, and usually I would inline the documentation. However in this case a lot of commands are very complicated and require a lot of documentation.
If you find a good way to do this, I would welcome a PR.
Here's the User Manual:
https://www.nxp.com/docs/en/user-guide/141520.pdf

closed 2024-03-04 by dimpolo