WMT-GmbH/pn532 #24: Pn532 reliability?

yanshay, 2024-12-15:
I find pn532 to not work reliably (at least with this library, haven't tried C libraries so can't say if it is only device issues or maybe software issue), is there something that can be configured for it to be more reliable?

Issues I face - 
- I needed to work with I2C ver low frequency (20kHz), while I read it should be able to work at MHz - which maybe make it more reliable in case the tag moves a bit while reading because it is faster?
- Quite a few times it doesn't recognize tag even when very close
- Many times it begins to read and fails after several pages, I do retries but many times they still fail, with error codes 1, 2 or 11. Even begin with some error, then switch to another
- It is very sensitive to distance of tag (sometimes tag is too near actually and need to be further away).
- Sometimes the device goes into an error state (while all I'm doing is reading/writing)

Are there any configurations available to tune the behavior?

Any thoughts about implementing this library also for pn7160 which I read is more reliable?

dimpolo, 2024-12-15:
Yeah, reliability is not great in my experience as well.
I've also seen the issues you describe come up with the exception of I2C frequency.

What are the errors you're getting when using 100 kHz I2C? With I2C cable length and pull-up resistor sizing can be issues. Maybe you can analyze your signals with a scope to check for noise and rise-times.

I've found that EMI and metal near the tag can lead to problems.

>Are there any configurations available to tune the behavior?

Unfortunately I don't know, if you figure something out, let me know :)

>Any thoughts about implementing this library also for pn7160 which I read is more reliable?

Does it use a similar protocol?

yanshay, 2024-12-15:
> > Any thoughts about implementing this library also for pn7160 which I read is more reliable?
>
> Does it use a similar protocol?

I don't know since I don't know the pn532 protocol either.
I guess it's different but its also by NXP so maybe there are similarities.
It's a replacement for pn532.
I read somewhere that pn532 manufacturing ended, don't know how reliable that information is.

This is the user-manual:
https://www.nxp.com/docs/en/user-manual/UM11495.pdf

yanshay, 2024-12-15:
> With I2C cable length and pull-up resistor sizing can be issues. Maybe you can analyze your signals with a scope to check for noise and rise-times.

Do I need to add resistors with I2C? I just connect it directly to Esp32 with around 20cm wires.

Is SPI more reliable and easier from that respect? I couldn't get it to work at all. Do you happen to have example code for esp32 that work with SPI?

dimpolo, 2024-12-15:
> This is the user-manual: https://www.nxp.com/docs/en/user-manual/UM11495.pdf

Looks very different to me, so I don't think there's much code that can be shared here.

> Do I need to add resistors with I2C? I just connect it directly to Esp32 with around 20cm wires.

Yes, absolutely. 4.7k or 10k are typical values. You might want to have a read online.

> Is SPI more reliable and easier from that respect? I couldn't get it to work at all. Do you happen to have example code for esp32 that work with SPI?

SPI tends to be more reliable in my experience. 

I believe your implementation from #22 wasn't quite correct. Since the cs handling is now part of the SpiDevice trait one needed to be sure, to stick bytes that belong together into a single transaction. Maybe you want to give SPI another try with #25

yanshay, 2024-12-15:
> Yes, absolutely. 4.7k or 10k are typical values. You might want to have a read online.

Doesn't the Esp32 have the pull-ups internally?

dimpolo, 2024-12-16:
Yes, but those are much larger than 10k

yanshay, 2024-12-16:
I gave SPI another try, couldn't get it to work (with esp32).
Below is the code, anything wrong with it? I get `TimeoutAck`.

I set the dip-switches on the module to spi.

```
    let sck = peripherals.GPIO14;
    let mosi = peripherals.GPIO13;
    let miso = peripherals.GPIO12;
    let cs = Output::new(peripherals.GPIO15, Level::High);

    let spi = Spi::new_with_config(
        peripherals.SPI2,
        esp_hal::spi::master::Config {
            frequency: 20.kHz(),
            mode: SpiMode::Mode0,
            ..esp_hal::spi::master::Config::default()
        },
    )
    .with_sck(sck)
    .with_mosi(mosi)
    .with_miso(miso);

    let spi_device =
        embedded_hal_bus::spi::ExclusiveDevice::new_no_delay(spi, cs).unwrap();

    let interface = pn532::spi::SPIInterface { spi: spi_device };

    let timer = TimerWrapper {
        timer: PeriodicTimer::new(timer0),
    };

    info!("Creating pn532");
    let mut pn532: Pn532<_, _, 32> = Pn532::new(interface, timer);

    info!("Configuring pn532");

    if let Err(e) = pn532.process(
        &Request::sam_configuration(SAMMode::Normal, false),
        0,
        500.millis(),
    ) {
        info!("Could not initialize PN532: {e:?}")
    }

```

yanshay, 2024-12-16:
Managed to get it working, I had to use the `features = ["msb-spi"]` for it to work.

dimpolo, 2024-12-17:
Awesome! Note that on the esp you can set the bit-order
https://docs.rs/esp-hal/latest/esp_hal/spi/master/struct.Config.html