WMT-GmbH/pn532 #27: Raspberry Pi 3B - Input / Output error

GChalony, 2024-12-31:
I'm trying to use the PN532 with a Raspberry Pi 3B. The PN532 is connected via I2C, and works well with `nfc-poll`. However, when I run the PN532 code (using rppal's hal implementation), I get a `Input/output error`. And this seems to "break" the I2C communication, because afterwards `nfc-poll` also returns a Timeout error. Even `i2cdetect -y 1` stops showing "24" (and scans really slowing compared to when it work ?).

It might be an issue with `rppal`, but I have the same issue with `linux-embedded-hal`. Did I misunderstand something in the setup code ?
Any help would be greatly appreciated :wink: 

Here's what I'm seeing:
```shell
test-hal-pn532 $ nfc-poll                            <<< POLLING WORKS FINE
nfc-poll uses libnfc 1.8.0
NFC reader: PN532 over I2C opened
NFC device will poll during 36000 ms (20 pollings of 300 ms for 6 modulations)
ISO/IEC 14443A (106 kbps) target:
    ATQA (SENS_RES): 00  44
       UID (NFCID1): 04  78  9b  0a  92  72  80
      SAK (SEL_RES): 00
Waiting for card removing...nfc_initiator_target_is_present: Target Released
done.
test-hal-pn532 $ cargo run                           <<<  RUNNING SAMPLE CODE
    Finished `dev` profile [unoptimized + debuginfo] target(s) in 0.27s
     Running `target/debug/test-hal`
Could not initialize PN532: InterfaceError(Io(Os { code: 5, kind: Uncategorized, message: "Input/output error" }))
pi@boombox:~/Documents/Code/test-hal-pn532 $ nfc-poll             <<< POLLING DOESN'T WORK ANYMORE
nfc-poll uses libnfc 1.8.0
error   libnfc.bus.i2c  Error: wrote only -1 bytes (10 expected) (Connection timed out).
error   libnfc.driver.pn532_i2c Failed to transmit data. Retries left: 2.
...
pn53x_check_communication: Input / Output Error
error   libnfc.bus.i2c  Error: wrote only -1 bytes (10 expected) (Connection timed out).
error   libnfc.driver.pn532_i2c Failed to transmit data. Retries left: 2.
...
nfc-poll: ERROR: Unable to open NFC device.
```

And the corresponding code:
```rust
use std::convert::Infallible;
use std::time::Duration;
use std::time::Instant;

use pn532::nb;
use pn532::CountDown;
use rppal::i2c::I2c;
use pn532::i2c::I2CInterface;
use pn532::Pn532;
use pn532::Request;
use pn532::requests::SAMMode;
use pn532::IntoDuration;

// Stolen from rppal's implementation
#[derive(Debug, Copy, Clone)]
struct Timer {
    start: Instant,
    duration: Duration,
}

impl Timer {
    pub fn new() -> Self {
        Self {
            start: Instant::now(),
            duration: Duration::from_micros(0),
        }
    }
}

impl Default for Timer {
    fn default() -> Self {
        Timer::new()
    }
}

impl CountDown for Timer {
    type Time = Duration;

    fn start<T>(&mut self, timeout: T)
    where
        T: Into<Self::Time>,
    {
        self.start = Instant::now();
        self.duration = timeout.into();
    }

    fn wait(&mut self) -> nb::Result<(), Infallible> {
        if self.start.elapsed() >= self.duration {
            Ok(())
        } else {
            Err(nb::Error::WouldBlock)
        }
    }
}

fn main() {
    let i2c = I2c::new().expect("Couldn't create I2C");
    let timer = Timer::new();
    let interface = I2CInterface {
        i2c: i2c
    };
    let mut pn532: Pn532<_, _, 32> = Pn532::new(interface, timer);
    if let Err(e) = pn532.process(&Request::sam_configuration(SAMMode::Normal, false), 0, 500.ms()){
        println!("Could not initialize PN532: {e:?}")
    }
    if let Ok(uid) = pn532.process(&Request::INLIST_ONE_ISO_A_TARGET, 7, 1000.ms()){
        let result = pn532.process(&Request::ntag_read(10), 17, 50.ms()).unwrap();
        if result[0] == 0x00 {
            println!("page 10: {:?}", &result[1..5]);
        }
    }
}
```

dimpolo, 2025-01-05:
Hi, thanks for investigating this!

Looks like rppal does not correctly implement the I2c trait. I've opened an issue: https://github.com/golemparts/rppal/issues/171

There used to be a similar issue in `linux-embedded-hal` (see https://github.com/WMT-GmbH/pn532/issues/19).
Are you using `linux-embedded-hal = "0.4.0"` and transitively `i2cdev = "0.6.1"`?

GChalony, 2025-01-07:
Thanks for your answer. I've checked again but I see the same behavior with `linux-embedded-hal = 0.4.0` (what I was using already). 
```shell
$ cargo run --example linux_pn532
Could not initialize PN532: InterfaceError(I2CError { err: Errno(5) })
```
_Code is similar to my previous comment but using `linux-embedded-hal` instead._

 However it seems it's relying on [`i2cdev` version 0.6.0](https://github.com/rust-embedded/linux-embedded-hal/blob/master/Cargo.toml#L30). Digging a little more, 0.6.1 is available on crates.io, but on Github only the [0.6.0 is mentioned](https://github.com/rust-embedded/rust-i2cdev/releases). Do you know what's going on ?

dimpolo, 2025-01-07:
Try running `cargo update -p i2cdev` that should pull in 0.6.1 😃 
Github Releases are a separate thing, guess the maintainers just forgot to update them

GChalony, 2025-01-08:
Thanks, it works to update to 0.6.1. However the bug is still here:
```shell
$ nfc-poll
nfc-poll uses libnfc 1.8.0
NFC reader: PN532 over I2C opened
NFC device will poll during 36000 ms (20 pollings of 300 ms for 6 modulations)
ISO/IEC 14443A (106 kbps) target:
    ATQA (SENS_RES): 00  44
       UID (NFCID1): 04  78  9b  0a  92  72  80
      SAK (SEL_RES): 00
Waiting for card removing...
^C

$ cargo run --example linux_pn532
    Finished `dev` profile [unoptimized + debuginfo] target(s) in 5.68s
     Running `target/debug/examples/linux_pn532`
Could not initialize PN532: InterfaceError(I2CError { err: Errno(5) })        <<< ERROR HERE

$ nfc-poll                                                                    <<< THEN BROKEN
nfc-poll uses libnfc 1.8.0
error   libnfc.bus.i2c  Error: wrote only -1 bytes (10 expected) (Connection timed out).
error   libnfc.driver.pn532_i2c Failed to transmit data. Retries left: 2.
error   libnfc.bus.i2c  Error: wrote only -1 bytes (10 expected) (Connection timed out).
error   libnfc.driver.pn532_i2c Failed to transmit data. Retries left: 1.
error   libnfc.bus.i2c  Error: wrote only -1 bytes (10 expected) (Connection timed out).
error   libnfc.driver.pn532_i2c Failed to transmit data. Retries left: 0.
error   libnfc.driver.pn532_i2c Unable to transmit data. (TX)
pn53x_check_communication: Input / Output Error
error   libnfc.bus.i2c  Error: wrote only -1 bytes (10 expected) (Connection timed out).
error   libnfc.driver.pn532_i2c Failed to transmit data. Retries left: 2.
error   libnfc.bus.i2c  Error: wrote only -1 bytes (10 expected) (Connection timed out).
error   libnfc.driver.pn532_i2c Failed to transmit data. Retries left: 1.

$ cargo tree -i i2cdev
i2cdev v0.6.1
└── linux-embedded-hal v0.4.0
    └── test-hal v0.1.0 (/home/pi/Documents/Code/test-hal-pn532)
```
`cargo tree` does report v0.6.1 for `i2cdev`.
Do you know if there's a way to inspect the I2C traffic to understand what's going on ?

dimpolo, 2025-01-08:
Hm interesting.
Does anything change when you run it with --release?

GChalony, 2025-01-09:
Nope same thing

GChalony, 2025-01-12:
Not sure if it's related, but I fried my USB ports. Though as `nfc-poll` works fine, I don't think it's really relevant. Do you have any idea how I can investigate ?

GChalony, 2025-01-30:
Hey @dimpolo, do you have any hint as to where to look for, to understand this bug?

dimpolo, 2025-01-31:
Sorry for not getting back at you, unfortunately I don't have much time for this repo.

Looking at the logs it looks like the request gets sent successfully. The first wait_ready call then fails with the I/O error.

This reminds me of this comment on another issue:
> Just adding this as it seems relevant: PN532 datasheet section "12.26 Timing for the I2C interface" lists a minimum time between a STOP condition and a START condition of 1.3ms...  It also states that it may need to stretch the clock when woken by up to 1ms. 

 _Originally posted by @liebman in [#4](https://github.com/WMT-GmbH/pn532/issues/4#issuecomment-2094834050)_ 

Could you see if inserting some delays before calls to wait_ready solves your issue? 
You can write your own implementation of `process` to do so https://github.com/WMT-GmbH/pn532/blob/master/src/protocol.rs#L152-L173

Ultimately this looks like a Linux/Raspberry issue to me. The whole bus shouldn't just break because the pn532 failed to respond to a message

GChalony, 2025-02-09:
So I've manage to reproduce the issue without this repo's code:
```rust
use std::{thread::sleep, time::Duration};

use rppal::i2c::I2c;

fn main() {
    let mut i2c = I2c::new().expect("Couldn't create I2c");
    i2c.set_slave_address(0x24).expect("Couldn't set address");

    // Get firmware version
    i2c
    .write(&[
        0x0, 0x0, 0xFF,  // Preamble
        0x02,  // Frame len
        0xFE,  // Checksum
        0xD4,  // Host to PN532
        0x02,  // Get Firmware Version command
        0x2A,  // Frame checksum
        0x00,  // Postamble
        ])
    .expect("Couldn't send get firmware");

    // Wait ready
    let mut buf = [0];
    loop {
        let r = i2c.read(&mut buf);
        match r {
            Err(e) => println!("Error reading for ready {}", e),
            Ok(_) => println!("Ready {}", buf[0])
        }
        if buf[0] == 1 { break };
        sleep(Duration::from_millis(1));   
    }

    // Receive ACK
    const ACK: [u8; 7] = [0x01 /* RDY*/, 0x00, 0x00, 0xFF, 0x00, 0xFF, 0x00];
    let mut ack_buf = [0; 7];
    i2c.read(&mut ack_buf).expect("Couldn't read ACK");
    if ack_buf != ACK {
        println!("Didn't receive ACK: {:?}", ack_buf);
    }

    // Wait ready
    let mut buf = [0];
    loop {
        let r = i2c.read(&mut buf);
        match r {
            Err(e) => println!("Error reading for ready {}", e),
            Ok(_) => println!("Ready {}", buf[0])
        }
        if buf[0] == 1 { break };
        sleep(Duration::from_millis(1));   
    }

    // Read response
    let mut buf: [u8; 13] = [0; 13];
    i2c.read(&mut buf).expect("Couldn't read resp");
    println!("Received response\n{:?}", buf);

}
```

Which gives:
```
Error reading for ready I/O error: Input/output error (os error 5)
Ready 1
Ready 0
Ready 1
Received response
[1, 0, 0, 255, 6, 250, 213, 3, 50, 1, 6, 7, 232]
```
The response is the correct one, IC Ox32, version 1.6.

So the "Input / Output error" seems to occur when we query the PN532 too fast, before it has time to process even acknowledge the request (as mentioned in #4).

There are still a few things that I don't understand :
 - Why the `pn532` code "breaks" the I2C line ? Once run (and failed), `nfc-poll` or any other way to communicate with the device (`I2c::write`) returns this "Input / Output error"
 - Can a well placed delay fix the issue ?
 - Isn't clock stretching supposed to handle this exact issue ?

It took me a moment but I finally noticed that the `I2cInterface::read` command discards the first byte, which is the RDY byte. Maybe a comment would be useful there ?

GChalony, 2025-02-09:
I tracked the bug a little further:

## Step 1: retry when not ready
In `I2c::wait_ready`, the initial Error received is of kind "Other", and therefore rethrown, instead of asking for a delay like with `NoAcknowledge`. I've changed the code it to always return `Pending`.

## Step 2: RDY byte received twice
This works as expected, `wait_ready` retries and eventually we receive READY = 1. However now I have a new error: `BadAck`. That's because the received buffer still contains the READY byte 
```
[1, 0, 0, 255, 0, 255]
```
That's probably because of https://github.com/golemparts/rppal/issues/171 ?

## Step 3: only one Read in transaction
To fix the issue, I thought I could change the `I2cInterface::read` to only perform one read, and then discard the first bytes and copy the rest.
```rust
let res = self.i2c.transaction(
    I2C_ADDRESS,
    &mut [Operation::Read(&mut self.buffer[..buf.len() + 1])]
);
println!("Read {:?}", self.buffer);
buf.copy_from_slice(&self.buffer[1..buf.len() + 1]);
```

It works! 
(Sorry the logging became a little verbose)
```
Sending: SAMConfiguration [1, 0, 0]
Waiting until ready
Err ready RDY: Io(Os { code: 5, kind: Uncategorized, message: "Input/output error" }) Other
Ready 1
Receiving ACK
Read (request len = 6) [1, 0, 0, 255, 0, 255, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]
Waiting until ready
Ready 1
Receiving response to SAMConfiguration
Receiving response of len 0
Read (request len = 9) [1, 0, 0, 255, 2, 254, 213, 21, 22, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]
Parsing response: [0, 0, 255, 2, 254, 213, 21, 22, 0]
Sending: InListPassiveTarget [1, 0]
Waiting until ready
Err ready RDY: Io(Os { code: 5, kind: Uncategorized, message: "Input/output error" }) Other
Ready 1
Receiving ACK
Read (request len = 6) [1, 0, 0, 255, 0, 255, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]
Waiting until ready
Ready 1
Receiving response to InListPassiveTarget
Receiving response of len 16
Read (request len = 25) [1, 0, 0, 255, 15, 241, 213, 75, 1, 1, 0, 68, 0, 7, 4, 146, 156, 10, 146, 114, 128, 211, 0, 67, 53, 129, 0, 0, 0, 0, 0, 0]
Parsing response: [0, 0, 255, 15, 241, 213, 75, 1, 1, 0, 68, 0, 7, 4, 146, 156, 10, 146, 114, 128, 211, 0, 67, 53, 129]
Tag UID: [1, 1, 0, 68, 0, 7, 4, 146, 156, 10, 146, 114, 128]
Done
```

---
So :
1. The first issue is the same as #4, the PN532 doesn't respond immediately. However `rppal` doesn't report it as a `NoAcknowledge`, so it fails instead of retrying. Could be fixed by retrying anyway, waiting beforehand, or catching the actual error.
2. Transaction protocol is not respected, and therefore the `I2CInterface` fails to strip the RDY byte at the beginning of each frame. Can be fixed upstream, or by sending just one READ instead of two
3. I still don't know why the I2C bus can become "broken" and break any conversation with the PN532, even with `nfc-poll`

@dimpolo would you give me your advice on all of that? I'd be happy to make a PR

dimpolo, 2025-02-10:
Thanks for sharing your findings!

I found this article claiming that RPi 3 does not support clock stretching.
https://learn.adafruit.com/raspberry-pi-i2c-clock-stretching-fixes/overview#why-are-raspberry-pis-an-issue-3130020

This linked issue claims that the RPi 5 solved that issue
https://github.com/raspberrypi/linux/issues/4884

In terms of what we can do here:

>It took me a moment but I finally noticed that the I2cInterface::read command discards the first byte, which is the RDY byte. Maybe a comment would be useful there ?

That sound like a good idea

> @dimpolo would you give me your advice on all of that? I'd be happy to make a PR

We could have a `rpi_compat` module behind a feature gate, with a separate `I2CInterface` implementation.
This would:
* perform a single read in wait_ready
* add a delay, that avoids I/O errors

GChalony, 2025-02-10:
Interesting! I didn't know about the issue with the Raspberry Pi. That probably explains why it becomes "broken" (i.e. out of sync, as described in the article you linked).

Regarding the `rpi_compat`, it's as you wish. I was able to fix it myself by simply using a custom implementation of  `Interface` so I'm happy with that. I'll also look into fixing the transaction protocol upstream.  
Tell me if you want me to make a PR on that.

dimpolo, 2025-02-11:
Sounds awesome!